ArduPilot/ardupilot_wiki #6302: Add "EK3 sources require xxx" to Arming Checks List

rmackay9, 2024-09-26:
We had a user that could not arm their vehicle because they got this message:

"EK3 source requires GPS".

The cause of the problem was that EK3_SRCx_YAW = 8 (GSF) and GSF currently requires GPS to work.  As a side note it should be possible to make GSF work using any position sensor (not just GPS) but this is another issue on the to-do list.

Anyway, we should list-up all the possible messages that can come from this part of the code https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF/AP_NavEKF_Source.cpp#L326

By the way, not trying to put monkeys on anyone's backs.  I can hopefully do this, I just don't want the issue to be forgotten

Hwurzburg, 2024-09-26:
Actually it wont be a lot of work...only two new error messages to the table....one for missing sensors and one for EKF source errors.....but I don't know the "solution" for the latter....solution text for former is easy..suggestions?

rmackay9, 2024-09-26:
Hi @Hwurzburg,

I think the solution is either to attach a GPS or change the EK3_SRCx_POSXY, EK3_SRCx_VELXY from "3" to something else and/or change EK3_SRCx_YAW is 8 then change that to something else.  Sorry, this is not the exactly best wording but something to this effect..

Hwurzburg, 2024-09-26:
I think you missed my intent...we can discuss tomorrow AM