ArduPilot/ardupilot_wiki #6311: ROS 2: microxrceddsgen version: null

AssaX0, 2024-10-01:
Hello,

currently I am trying to get a Pixhawk6C to raspberrypi4B connection Working.
I am using Ubuntu 22.04 with ROS2 Humble and try to follow the Tutorial:
[ROS 2](https://ardupilot.org/dev/docs/ros2.html) 
I followed everything (twice - removed everything and tried again after it did not work) up to the point where you are supposed to check the
"microxrceddsgen -version"
wich returns
"microxrceddsgenversion: null"
I tried fixing it myself, but I am no Linux Main so I asked chat gpt who found a probable typo (?) in:
nano /home/pi/ardu_ws/Micro-XRCE-DDS-Gen/scripts/microxrceddsgen
one Line says
"if ! hash ${jave_exec}; then" what should probably be a:
"if ! hash ${java_exec}; then" ... but after fixing I stil got a 
"microxrceddsgen version: null" and chatgpt is not helpfull anymore.

I hope someone can help me fixing this!

Ryanf55, 2024-10-15:
Ignore the version. We can't do a formal release.

Use `microxrceddsgen -h` to check it works.

AssaX0, 2024-10-18:
Well i guess it doesnt?
[image]

And If I would like to continue the dds tests start to fail... I guess thats the consequence?
[image]

Ryanf55, 2024-10-19:
> Well i guess it doesnt? [...]

I put a PR up to solve the version printout. Please give it a try if you can and provide a review.

As far as why your tests are failing, please ask on discuss on one of the ROS threads. I'll need to to run it with more verbosity and attach logs. The ardupilot wiki issues is for discussing issues with docs, not user support.

Ryanf55, 2024-10-27:
@Hwurzburg Please close this when convenient. It's complete.

closed 2024-10-27 by Hwurzburg