ArduPilot/ardupilot_wiki #6316: Vagrant doesn't work on Apple Silicon Macs because ubuntu/jammy64 is not an Arm image

master0v, 2024-10-03:
I am following the instructions on this page: https://ardupilot.org/dev/docs/setting-up-sitl-using-vagrant.html

When I get to "vagrant up" step - it craps out due to the fact that Vagrant 2.4.1 doesn't support virtualbox Version 7.1.2 r164945 (Qt6.5.3).  That can be fixed with the suggestion provided here: https://github.com/hashicorp/vagrant/issues/13501

Once that is working, however, the fundamental problem is that the VM that it is trying to pull is built for the Intel architecture.  Here is the full error message:

```
MacBook-Pro ardupilot % vagrant up --provider=virtualbox   
Bringing machine 'jammy' up with 'virtualbox' provider...
==> jammy: Checking if box 'ubuntu/jammy64' version '20240912.0.0' is up to date...
==> jammy: Clearing any previously set forwarded ports...
==> jammy: Clearing any previously set network interfaces...
==> jammy: Preparing network interfaces based on configuration...
    jammy: Adapter 1: nat
==> jammy: Forwarding ports...
    jammy: 22 (guest) => 2222 (host) (adapter 1)
==> jammy: Running 'pre-boot' VM customizations...
==> jammy: Booting VM...
There was an error while executing `VBoxManage`, a CLI used by Vagrant
for controlling VirtualBox. The command and stderr is shown below.

Command: ["startvm", "8ef543d5-204a-4df4-830f-d075932bdcd3", "--type", "headless"]

Stderr: VBoxManage: error: The VM session was aborted
VBoxManage: error: Details: code NS_ERROR_FAILURE (0x80004005), component SessionMachine, interface ISession

```

The same issue happens when I try to run the VM from the UI:

[image]

Would it be possible to support this?
Thank you!

peterbarker, 2024-10-03:
From random googling, it may be possible by changing the machine from eg. ubuntu/jammy64 to  perk/ubuntu-2204-arm64

https://portal.cloud.hashicorp.com/vagrant/discover/perk/ubuntu-2204-arm64

There may also be bento boxes to allow for it.

You might need to experiment a bit :-)

master0v, 2024-10-03:
Forgive me for the dumb question, but how do you change it?

Simply replacing 'ubuntu/jammy64' with 'perk/ubuntu-2204-arm64' in `Vagrantfile` didn't work:

```
MacBook-Pro ardupilot % virtualbox
2024-10-03 16:59:04.293 VirtualBox[91865:1039886] +[IMKClient subclass]: chose IMKClient_Legacy
2024-10-03 16:59:04.293 VirtualBox[91865:1039886] +[IMKInputSession subclass]: chose IMKInputSession_Legacy
aim@Alexeis-MacBook-Pro ardupilot % vagrant up --provider=virtualbox
Bringing machine 'jammy' up with 'virtualbox' provider...
==> jammy: Box 'perk/ubuntu-2204-arm64' could not be found. Attempting to find and install...
    jammy: Box Provider: virtualbox
    jammy: Box Version: >= 0
==> jammy: Loading metadata for box 'perk/ubuntu-2204-arm64'
    jammy: URL: https://vagrantcloud.com/api/v2/vagrant/perk/ubuntu-2204-arm64
The box you're attempting to add doesn't support the provider
you requested. Please find an alternate box or use an alternate
provider. Double-check your requested provider to verify you didn't
simply misspell it.

If you're adding a box from HashiCorp's Vagrant Cloud, make sure the box is
released.

Name: perk/ubuntu-2204-arm64
Address: https://vagrantcloud.com/api/v2/vagrant/perk/ubuntu-2204-arm64
Requested provider: virtualbox (arm64)

```

Knowledgelocater, 2024-11-13:
Add the following to your Vagrantfile before running vagrant up:

ruby
Vagrant.configure("2") do |config|
  config.vm.provider "virtualbox" do |vb|
    vb.customize ["modifyvm", :id, "--cpuidset", "00000001", "000306a9", "00100800", "7ffafbbf", "bfebfbff"]
  end
end

Update vagrant
bash:-  
brew install vagrant

otherwise use this :-
Vagrant.configure("2") do |config|
  config.vm.box = "bento/ubuntu-22.04-arm64"  # Choose an ARM-compatible box
end
 

and re initialize 
vagrant reload --provision

jeromecovington, 2024-12-20:
I am finding that `bento/ubuntu-22.04-arm64` is not compatible with Virtualbox as the provider. Has anyone else experienced differently?

xooooooooox, 2024-12-28:
I have the same problem

Knowledgelocater, 2024-12-29:
if you guys are having problem with this bento/ubuntu-22.04-arm64 then directly use bento/ubuntu-22.04 it is compatible.
For use :- vagrant init bento/ubuntu-22.04
                vagrant up --provider=virtualbox
as this is compatible on x86_64 if not this also the use generic.

I hope this will be helpfull

Knowledgelocater, 2025-01-09:
@mknet if there is still problem then , i will check it and inform you? 
Thanks for letting me know!

mknet, 2025-01-10:
> @mknet if there is still problem then , i will check it and inform you? Thanks for letting me know!

Yes, @Knowledgelocater, the problem still remains. Any hints?

GDivino, 2025-01-22:
If you are using macbook M series chips, this is due to the incompatibility of the boxes you are choosing with your provider and architecture.

Assuming you are using virtualbox with macbook M series chips, you need to look for these types of boxes:
- Architecture: arm64
- Provider: Virtualbox
[image]

You can test out [this box](https://portal.cloud.hashicorp.com/vagrant/discover/net9/ubuntu-24.04-arm64). It worked for me with virtualbox.

Do note: Virtualbox has very few official vagrant boxes that support ARM64, so if you want the ability to use many, try switching to the qemu provider.

jeromecovington, 2025-01-22:
@GDivino yes, I came across that one. Was really hoping to find one by a more well known maintainer before I use it at work. But, thanks for restating the problem. It seems to me that it was only partially understood based on some of the comments on this issue.

jainsiddharth99, 2025-06-25:
This worked for me 
https://portal.cloud.hashicorp.com/vagrant/discover/bento/ubuntu-22.04

try this 
```
unifiedcontroller.vm.box = "bento/ubuntu-22.04"
unifiedcontroller.vm.box_version = "202502.21.0" 
unifiedcontroller.vm.box_architecture = "arm64"
```

peterbarker, 2025-06-25:
> ```
> unifiedcontroller.vm.box = "bento/ubuntu-22.04"
> unifiedcontroller.vm.box_version = "202502.21.0" 
> unifiedcontroller.vm.box_architecture = "arm64"
> ```

Any chance of a (tested) PR?

peterbarker, 2025-10-05:
> Although late, I just wanted to add my working vagrant config for an M4 Mac running 15.6.1 based on this thread:
> 
> ```
>   config.vm.define "noble", primary: true do |noble|
>     noble.vm.box = "bento/ubuntu-22.04"
>     noble.vm.box_version = "202508.03.0"
>     noble.vm.provision :shell, path: "Tools/vagrant/initvagrant.sh"
>     noble.vm.provider "virtualbox" do |vb|
>       vb.name = "ArduPilot (noble)"
>     end
>     noble.vm.boot_timeout = 1200
>   end
> ```
> 
> As a further FYI, once the VM was running, I couldn't get Mavproxy to forward out of the VM to the Mac host over UDP. So I ended up starting the `sim_vehicle.py` with `--no-mavproxy` switch, and configured VirtualBox to port forward `tcp:5760` and ran MavProxy on my MacOS host: `mavproxy.py --master=tcp:127.0.0.1:5760 --out=udp:0.0.0.0:14550`. QGC then connected immediately.

Could you create a PR against master for that, please?  Given the version pin it might be a good idea to use `noble-mac-m4` for the machine name?