ArduPilot/ardupilot_wiki #6336: Gazebo 11 not working on ubuntu 22 and 24

M-Usamah, 2024-10-17:
I am using ubuntu24.04 lts and trying to install ardupilot with Gazebo11 but it is not installing and having this issue 

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package gazebo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gazebo' has no installation candidate

how to fix this issue

Ryanf55, 2024-10-17:
Gazebo 11 is not supported on Ubuntu 24. It's almost end of life. You need to move to the new gazebo (harmonic, or similar).
https://gazebosim.org/docs/harmonic/getstarted/

Ryanf55, 2024-10-17:
Where did the wiki tell you to install Gazebo 11 on Ubuntu 24?

M-Usamah, 2024-10-17:
no but i am using ubuntu 24 as my primary system

so how to communicate or connect it with ardupilot

Ryanf55, 2024-10-17:
@srmainwaring and I have not fully vetted Ubuntu 24 yet for ROS development. Once we have, it can be added to the wiki.

I would recommend trying:
* Uninstall all current versions of Gazebo
* Install gazebo ionic on ubuntu 24 (This is the LTS version recommended for Ubuntu 24). 
* Run some of the gazebo examples without ROS or Ardupilot
* Set up ardupilot using the setup script: https://ardupilot.org/dev/docs/building-setup-linux.html#install-some-required-packages
* Clone this repo: https://github.com/ArduPilot/ardupilot_gazebo
* It will likely require some modifications to work on Ubuntu 24. If you can, submit a PR for your changes.

srmainwaring, 2024-10-17:
> I am using ubuntu24.04 lts and trying to install ardupilot with Gazebo11

We don't support Gazebo 11 as it conflicts with Gazebo Garden and later and is EOL January 2025.