ArduPilot/ardupilot_wiki #6369: FLIR Vue Pro

sarathkumar56c, 2024-11-05:
[FLIR Vue Pro](https://ardupilot.org/copter/docs/common-flir-vue-pro.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

sarathkumar56c, 2024-11-05:
i'm using flir duo r 
im trying to trigger via mavlink using pixhawk 3
i cant able to sync GPS data in the images

Hwurzburg, 2024-11-05:
you should post this support question in discuss.ardupilot.org, maybe in hardware section or copter section.....wont get much attention here since this is only for documentation issues

closed 2024-11-05 by Hwurzburg

rmackay9, 2024-11-06:
@sarathkumar56c,

As @Hwurzburg says this isn't the right place to discuss but in any case, I've used the Flir Vue Pro in the Japan Innovation Challenge but found that, like you, I also couldn't get the EXIF to work any more.  I suspected that it was some change in the NMEA output feature in ArduPilot ([the feature is briefly mentioned here](https://ardupilot.org/copter/docs/common-serial-options.html#serialx-protocol)) but I ran out of time to investigate and we eventually gave up on the camera and moved to the much more capable Siyi ZT6.  See blog post here https://discuss.ardupilot.org/t/tap-j-at-the-japan-innovation-challenge-2024/125750

sarathkumar56c, 2024-11-06:
thanks for the response

sarathkumar56c, 2024-11-06:
okey i will post it there