ArduPilot/ardupilot_wiki #6414: GPS for Yaw (aka Moving Baseline)

stannachev, 2024-12-04:
Hello!

Currently I have installed 2 GPS - Here 4 with Orange Plus cube and we are facing problem with getting Fixed RTK and there is a error message: "Prearm: Need Position Estimate" and in the EKF screen I have "pos_horiz_abs OFF" in red. Attached I`m sending our settings.
Any suggestions are more than welcome!

Hwurzburg, 2024-12-04:
this is not a wiki issue....support issues should be raised in discuss.ardupilot.org to on the Discord help threads...not many devs look at wiki issues....will close this

stannachev, 2024-12-04:
Thank you for the very fast response!

 

I`ll write there directly. 

 

Thank you for the support!

 

Best Regards,

 

Stanimir Nachev

General Manager

Mobile: +359 878 47 87 64

e-mail:  ***@***.***> ***@***.***

Web: www.aerovision.bg