ArduPilot/ardupilot_wiki #6436: MAVLink Sensor Bus Address

jordancormack, 2024-12-10:
Rangefinder and proximity sensors have a bus address parameter (e.g. [RNGFND1_ADDR](https://ardupilot.org/copter/docs/parameters.html#rngfnd1-addr-bus-address-of-sensor) which the docs say is used for I2C and DroneCAN sensors. 

I think MAVLink should also be added to the descrtiption for those parameters? When sending/reading a [DISTANCE_SENSOR](https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR) MAVLink message, the 'id' has to match the bus address for it to show up (similar to I2C and DroneCAN).

I realise that MAVLink rangefinders/proximity sensors are less common, but the wording for the bus address parameter(s) could still include MAVLink, or at least be written in a way to suggest it's not _just_ I2C and DroneCAN.

Hwurzburg, 2024-12-10:
that is a param relating ONLY to sensors attached to the autopilot....has nothing directly to do with MAVLink rangefinders and will not impact them at all..sorry have to disagree with this needing a change

the MAVLink message itself may or may not be impacted by the param

closed 2024-12-10 by Hwurzburg

jordancormack, 2024-12-10:
So if I have two MAVLink rangefinders, how do I assign one to RNGFND1 and the other to RNGFND2?

Is that not what the _ADDR parameter is intended to be for?

Hwurzburg, 2024-12-10:
no only I2c bus addrs of device attached to bus on autopilot
two mavlink would have RNGFND1_TYPE = MAVLink as well as RNGFND2_TYPE = MAVlink....**if** that is possible....its possible that there can only be only instanciation of the MAVLink RF backend....
you can ask in a support forum (discuss.ardypilot.org) for confirmation

rmackay9, 2024-12-10:
Hi @jordancormack,

The DISTANCE_SENSOR's orientation field is compared to the RNGFNDx_ORIENT parameter and if they match then the distance will be consumed.  Can the mavlink enabled rangefinder's orient field be configured?

jordancormack, 2024-12-11:
So it's just the orientation that has to match?

Meaning multiple MAVLink rangefinders with the same RNGFNDx_ORIENT will always show the same value (the latest DISTANCE_SENSOR message with a matching orientation), even if coming from separate sources with separate IDs?

Is that due to Ardupilot only being able to use one rangefinder 'per orientation'? Or is it specific to MAVLink rangefinders?

I am doing some testing with multiple MAVLink rangefinders and was hoping they would show up separately (e.g. one as rangefinder1 and the other as rangefinder2) rather than 'overwriting' each other.

rmackay9, 2024-12-11:
Hi @jordancormack,

Yes, based on [the code here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_MAVLink.cpp#L31) I think only the orientation field has to match the RNGFNDx_ORIENT.

AP can have multiple rangefinders setup pointing in the same direction but only the first healthy one will be used (e.g. RNGFND1 will be used before RNGFND2).  I guess that the distances from both would be sent to the GCS though.

We could potentially change the condition (linked above) so that it uses the "id" of the DISTANCE_SENSOR message instead of the orientation.  If you're comfortable compiling AP yourself you could probably make that change yourself and try it out.  Worst case I guess I could make the change for you but ideally I'd like to find a solution that works for everyone

jordancormack, 2024-12-11:
Yes it seems like the code would need to be changed a bit to get things playing nicely, and that'd probably need a discussion to figure out the best solution for everyone as you say.

I was expecting/hoping for a setup similar to some other sensors, e.g. CAN battery monitors with BATT_SERIAL_NUM that can be set to -1 or set to match a specific battery ID if you have more than one and want to know which is which.

In the short term for monitoring multiple MAVLink rangefinders I will probably use pymavlink or Mission Planner's MAVLink Inspector (changing the component IDs should make them show up separatly on there).

Then in the longer term I will play with that code you shared to see what works and makes sense.

Thanks both for the help!

rmackay9, 2024-12-11:
Hi @jordancormack,

yes, that makes sense.  I've created an [issue here](https://github.com/ArduPilot/ardupilot/issues/28837).  No promises when it will be done but it's a good idea, thanks!