ArduPilot/ardupilot_wiki #6595: Cartographer SLAM with ROS 2 in SITL

BilalBabar-edgec, 2025-02-04:
[Cartographer SLAM with ROS 2 in SITL](https://ardupilot.org/dev/docs/ros2-cartographer-slam.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Followed teh guide on this page, in the ardupilot configuration part I set the parameters as shown but still i get the error 
[ruby $(which gz) sim-1] [Wrn] [ArduPilotPlugin.cc:1583] ArduPilot controller has reset
[mavproxy.py -5] AP: PreArm: Need Position Estimate
[mavproxy.py -5] AP: PreArm: VisOdom: not healthy
[mavproxy.py -5] AP: PreArm: Need Position Estimate
[mavproxy.py -5] AP: PreArm: VisOdom: not healthy

BilalBabar-edgec, 2025-02-05:
https://ardupilot.org/dev/docs/ros2-cartographer-slam.html
Followed the exact guide on this page, I was able to navigate the copter. When i configured the parameters the I got this issue where my copter couldn't takeoff or is not able to arm. 

[image]

BilalBabar-edgec, 2025-02-10:
> This may be a support issue which should be posted in discuss.ardupilot.org, but will leave it here until Ryan says it's not a wiki issue

Hey @Hwurzburg my query is not answered yet

Ryanf55, 2025-02-10:
I'm on vacation for another week. I'll get back to you when I get home and have a computer to test it on

Hwurzburg, 2025-03-10:
> I'm on vacation for another week. I'll get back to you when I get home and have a computer to test it on

@Ryanf55  ping

Ryanf55, 2025-03-10:
Please close this. It's a bug, not a wiki issue. 

Please follow this issue instead. The wiki probably won't need any changes.
https://github.com/ArduPilot/ardupilot_ros/issues/24