ArduPilot/ardupilot_wiki #6715: Dev: add link to lua script on "Requesting Data from Autopilot" page

rmackay9, 2025-03-24:
We should add a link to [this lua script](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/message_interval.lua) from [this developer wiki page](https://ardupilot.org/dev/docs/mavlink-requesting-data.html).

I was helping a user earlier today and they were looking for ways to reduce the bandwidth over a telemetry radio and I found this dev page was the best description we have except it's missing the Lua script method which I think is actually one of the the easiest way for users with this problem

Hwurzburg, 2025-03-25:
the easiest way is to stop a GCS from changing the rates an set then as params....setting up scripting (and its not possible in F4s) is more steps and complex

rmackay9, 2025-03-26:
Hi @Hwurzburg,

The solution really depends on what the user needs.  Anyway, l've linked a PR which I think gives more options to the user

Hwurzburg, 2025-03-26:
addressed by #6719

closed 2025-03-26 by Hwurzburg