ArduPilot/ardupilot_wiki #6718: Camara/Mount: explain how a mount's SysID mode can be used

rmackay9, 2025-03-25:
We recevied a message from a user in the pymavlink channel asking about how to make a camera mount follow a vehicle as shown in @timtuxworth's video linked below.
https://www.youtube.com/watch?v=FX60mqi9WMQ

We have a single line explaining this mode on the [Gimbal/Mount Controls page](https://ardupilot.org/copter/docs/common-mount-targeting.html) but we should probably add a dedicated page or at least a dedicated section explaining the setup.

Ashishbsharma, 2025-04-01:
I’ve created a pull request that adds a detailed section to the documentation explaining how to set up a gimbal to follow a vehicle using MAVLink MOUNT_TARGET, including parameter setup, a pymavlink example, and troubleshooting tips. This should help users trying to replicate setups like the one shown in @timtuxworth’s video. Please check.

rmackay9, 2025-04-01:
@Ashishbsharma, thanks for that.  I've put comments in the PR