ArduPilot/ardupilot_wiki #6764: Explain how to build bootloader with SD card firmware update feature included

rmackay9, 2025-04-16:
This page explains how to update an autopilot's firmware via MAVFTP and SD card but it relies on the .abin file being available for the autopilot on firmware.ardupilot.org

- https://ardupilot.org/copter/docs/common-install-sdcard.html

We should add instructions to this page or perhaps to https://ardupilot.org/dev/docs/bootloader.html to explain how a bootloader for any STM32 based autopilot can be built that includes this feature

Hwurzburg, 2025-04-17:
@peterbarker needs a couple of simple bullets and I will flesh out and add to wiki

rmackay9, 2025-04-21:
Here's a PR that appears to do what we want to explain https://github.com/ArduPilot/ardupilot/pull/29810

closed 2025-04-25 by Hwurzburg