ArduPilot/ardupilot_wiki #6797: Document how to lock-down parameters

rmackay9, 2025-04-24:
We occasionally see requests on the forums ([like this one](https://discuss.ardupilot.org/t/prevent-access-to-parameters-and-configuration-from-fc/109545)) asking how to lock-down parameters against accidentally being changed by the user.

AP-4.4 (and earlier) had the [APJ tool](https://ardupilot.org/dev/docs/apjtools-intro.html) which could be used for this purpose (I think) but it's not clear how to do it in 4.5 (and higher).  We should document the process.  If it's not possible then we should add the ability (which would be an autopilot or build environment enhancement).

aditi25srivastava, 2025-08-08:
great...let me do work on it . I'll inform you very soon after solving it!!!

deepak61296, 2025-12-28:
Hi @rmackay9 , looks like this was addressed in commit 726e5f9 (July 2025) - the @READONLY feature is now documented in the OEM customizations page: https://ardupilot.org/dev/docs/common-oem-customizations.html

does this cover what was needed, or is there more to add?