ArduPilot/ardupilot_wiki #6880: Rover: expand crash check to include crash angle

rmackay9, 2025-05-21:
Our Rover crash check will also trigger if the vehicle flips over so we should document this here. https://ardupilot.org/rover/docs/rover-failsafes.html#crash-check

Here's the parameter https://ardupilot.org/rover/docs/parameters.html#crash-angle-crash-angle

Related issue: https://github.com/ArduPilot/ardupilot/issues/29442

Hwurzburg, 2025-05-22:
does the angle limit exceed need to be for 2 seconds also before FS activation?

rmackay9, 2025-05-22:
@Hwurzburg,

No, it doesn't to trigger immediately.  Seems like it should need to wait for 2 seconds to be consistent but that's not how it's written

https://github.com/ArduPilot/ardupilot/blob/master/Rover/crash_check.cpp#L11

closed 2025-05-22 by Hwurzburg