ArduPilot/ardupilot_wiki #6925: DO_CHANGE_SPEED Type parameter's description inconsistent with MAVLink specification

horiuchi0photo, 2025-06-10:
[Copter Mission Command List](https://ardupilot.org/copter/docs/mission-command-list.html)
[Do-Change-Speed](https://ardupilot.org/copter/docs/mission-command-list.html#do-change-speed)

The description of the `Type` parameter for `DO_CHANGE_SPEED` does not seem to match the MAVLink specification.

Description says:

> 0:ground speed, 1:airspeed(if available) 

In Specification:
```xml
    <enum name="SPEED_TYPE">
      <description>Speed setpoint types used in MAV_CMD_DO_CHANGE_SPEED</description>
      <entry value="0" name="SPEED_TYPE_AIRSPEED">
        <description>Airspeed</description>
      </entry>
      <entry value="1" name="SPEED_TYPE_GROUNDSPEED">
        <description>Groundspeed</description>
      </entry>
      <entry value="2" name="SPEED_TYPE_CLIMB_SPEED">
        <description>Climb speed</description>
      </entry>
      <entry value="3" name="SPEED_TYPE_DESCENT_SPEED">
        <description>Descent speed</description>
      </entry>
    </enum>
```

rmackay9, 2025-06-10:
@horiuchi0photo, it seems you are right! thanks!

Hwurzburg, 2025-06-10:
fixed