ArduPilot/ardupilot_wiki #7018: FBWB_ELEV_REV wrong

734560602, 2025-07-15:
FBWB_ELEV_REV: Fly By Wire elevator reverse
Reverse sense of elevator in FBWB and CRUISE modes. When set to 0 up elevator (pulling back on the stick) means to lower altitude. When set to 1, up elevator means to raise altitude.

 When set to 0 up elevator (pulling back on the stick) means to raise altitude. 
Is this right?

closed 2025-07-15 by Hwurzburg

734560602, 2025-07-16:
Yes, it's  right  .but in ardupilot Complete Parameter List,
"When set to 0 up elevator (pulling back on the stick) means to lower altitude"

I think this description might be reversed.

Hwurzburg, 2025-07-16:
the param metadat is wrong...I have a PR submitted to the code repo to fix https://github.com/ArduPilot/ardupilot/pull/30667